goburrow/modbus #11: Wait until timeout on error

nemtsevp, 2016-05-22:
Hello. Can not receive proper error message. Trying to read non-existing address from code below. Code is running infinite if no timeout defined. Slave device is freemodbus-1.5.0 driven avr, was successfully tested with this https://sourceforge.net/projects/qmodmaster/

`package main

import (
	"github.com/goburrow/modbus"
	"fmt"
	"time"
)

func main() {
	handler := modbus.NewRTUClientHandler("/dev/ttyUSB0")
	handler.BaudRate = 115200
	handler.DataBits = 8
	handler.Parity = "N"
	handler.StopBits = 1
	handler.SlaveId = 1
	handler.Timeout = 10 * time.Second

	err := handler.Connect()
	fmt.Println(err)

	client := modbus.NewClient(handler)
	results, err := client.ReadInputRegisters(997, 1)

	fmt.Println(err)
	fmt.Println(results)
}`

nqv, 2016-05-23:
I'll try again, would you please let me know your OS? @nemtsevp

nemtsevp, 2016-05-23:
I've tried this on Ubuntu-16.04 @nqv

adiclepcea, 2016-05-30:
If you mean that you are trying to request data from a slave that does not exist, then it would be ok to wait until the timeout has exceeded as no response will arrive back. This implementation is not exactly following the minimum timeouts for package responses as defined by the modbus standard which is a minimum of 3.5 characters timeframe (depends on the speed). 
Most master implementations use a timeout of ~100ms (see. (http://modbus.control.com/thread/1262876422#1262876422)[here] answer 3 for example.
This implementation lets you decide how long to wait for a reply (hence the timeout).
This is a great feature in my opinion as there are loads of slaves that will not respect the 3.5 characters timing. You should have a timeout set however.

However, as in my PR, the implementation does not have to wait until the timeout has passed to see an error response.

nemtsevp, 2016-05-30:
I meant non-existing register address, device itself is visible, slave address is correct and correct register addresses are readable. Ok, i can set timeout and get a response, but even if timeout is set response is that timeout is reached, when it should be like "Invalid register address". So i can not ever get "Invalid register address" error. @adiclepcea Thanks for your participation! I think, your pull request is exactly what i need.

adiclepcea, 2016-05-30:
I see that the PR was accepted.
I am really glad to give a hand. Even more so since this project is a help for me too.

closed 2016-05-30 by nemtsevp