goburrow/modbus #13: Feature Request: Modbus slave simulator

fleimgruber, 2016-09-25:
Have you already thought about implementing a Modbus slave (TCP server)?

This would be a prerequisite for the following special use-case: say a certain slave device (lets call it "remote") only works in single-master mode (meaning only one master connected at any time). To connect it to multiple masters, one could create a "proxy" consisting of a slave and a master. The proxy slave forwards requests to the proxy master which in turn requests from the remote. The remote response is then returned to the masters via the proxy slave.

Another use-case would be to mask a Modbus RTU-only device with a TCP proxy to make it more widely available.

I would like to help out where I can after making myself more familiar with the code base.

nqv, 2016-09-25:
Yeah, I had thought about implementing a modbus server, but found all I need with [this binary](http://www.modbusdriver.com/diagslave.html).

But you're right, a proxy/router is a nice feature to have. Feel free to send a PR. I also will work on this if time permits.

Thank you

xiegeo, 2016-10-08:
Hi I am implementing a [Modbus library](https://github.com/xiegeo/modbusone) that also supports server/slave side, because I need to be able to switch between server and client modes. I am borrowing source code from here as much as possible, assuming @nqv is OK with that. 

The server is like a client turned inside out, and I like to experiment, so my code is too different to send any pull requests here, but I want to collaborate as much as possible.

OrangeTux, 2016-11-04:
@fleimgruber

OrangeTux, 2016-11-04:
> Another use-case would be to mask a Modbus RTU-only device with a TCP proxy to make it more widely available.

@fleimgruber  FWIW I had the same use case a while ago and build the solution in Python using. I've you are interested I could publish the source.

wuxc, 2019-05-10:
check this out [https://github.com/tbrandon/mbserver](mbserver)