goburrow/modbus #17: Managing multiple slaves with the RTU API, perhaps concurrently.

jonseymour, 2017-03-01:
For a use case I have I may need to talk to multiple slaves across a single RTU serial connection. It seems that the way to cater for this case is that to update RTUClientHandler.SlaveId immediately prior to each IO so that the packets are always encoded with the correct slave id.

Is my reading of the code correct?

I could do this in my application code, but then I have to pass around the RTUClientHandler object as well as the Client which isn't something I was planning to do.

I might also want to handle concurrent access to the underlying handler from multiple goroutines. Again, I can add locks in my code to ensure this happens.

However, I am wondering if there is case for a per-slave proxy of the Client interface which transparently switches the slaveId on the underlying handler and acquires a shared lock prior to invoking the underlying Client. My application would then have one Client for each slave and they would all safely share the underlying RTUClientHandler.

I will probably add this kind of Client proxy to my own application, but if you think the idea would be useful to other users of your API let me know and I'll submit a PR. If I am fundamentally misunderstanding something about how the API is intended to be used, let me know that too!

nqv, 2017-03-01:
wrt concurrent access, I'm in the process of adding mutex in the client transporter and fix #16 (expected to be done this week :) )
The client api doesn't have SlaveId in the parameter as using slaveId for TPC is rare. My idea originally is the ability to share the transporter layer among clients, which is like:
```
h := NewRTUClientHandler()
// Default client
c := NewClient(h)
// Create new Client for different SlaveId
h2 := NewRTUClientHandler()
h2.SlaveId = 2
c2 := NewClient2(h2, h)
```

May be I can add an helper
```
func NewClientWithSlaveId(c Client, slaveId int) Client {
   // Clone internal packager, set new slaveId and return a new Client with that packager
}
```
Please let me know if that will resolve your case. (Again, that requires #16 to fixed)

jonseymour, 2017-03-01:
I think from my point of view, `NewClientWithSlaveId(c Client, slaveId int)` does what I need and has the advantage that if I need to defer the creation of clients for other slaves, that I can do so without carrying the handler around.

jonseymour, 2017-03-01:
BTW: prototyping this, I realised that there is probably a need to add a `Close() `method to the Client interface and also a reference counter in the implementation to allow a Close() on the underlying handler to be deferred until there are no more references (otherwise this will have to be tracked separately).

nqv, 2017-03-02:
I don't think Close is needed. Will send a PR for #16 this weekend

gonium, 2017-07-19:
I had the same issue - which I solved by setting the ``Slave`` of the RTUClient before sending anything:

https://github.com/gonium/gosdm630/blob/master/src/github.com/gonium/gosdm630/modbus.go#L114

There is no need to ``Close()`` anything. I understand (and appreciate!) @nqv's view to provide a unified API independent of the transport, but for practical applications of the RTU transport, one needs to have a way to set the slave id.

HTH,
-Mathias

Aditya23456, 2018-04-04:
Hi,

Thanks for the great library. I don't see any development branch and was wondering if there's any support for concurrent Modbus client or any work done for supporting it ? 

Thanks once again,
Aditya