goburrow/modbus #18: Change input of WriteSingleCoil from uint16 to bool

cblauvelt, 2017-03-04:
Having the second input to Client.WriteSingleCoil  be a boolean instead of a uint16 that requires either 0xFF00 or 0x0000 seems more intuitive. What would you think of this change?

nqv, 2017-03-06:
Thanks for your suggestion. I agree that a boolean may be better in this case, but I also would like to not change the API to make it backward compatible. I think `uint16` is fine as that is what server responds. What do you think if I add `ON` and `OFF` constants (0xFF00 and 0x0000) instead?

cblauvelt, 2017-03-09:
I think that would work fine. I was thinking of defining these in my own program as well but if it's in the library it wouldn't be necessary. I would also recommend that there be an example in the README.