goburrow/modbus #20: Found a bug when slave device return error code

ggenien, 2017-05-09:
Modbus RTU Client:
**"panic: runtime error: slice bounds out of range"**

This panic occurs when slave device return an error code, and the frame return may be not standard.
source: rtuclient.go, n1
line: err = io.ReadFull(mb.port, data[n:5])

I changed for temporary:
err = fmt.Errorf("modbus: return error code")
n1 = 0

Maybe you could check code for similar problems.
Thank you for your product.

ggenien

nqv, 2017-05-12:
Thanks for your report, I'll take a look this weekend

gonium, 2017-07-19:
I'm running into the same problem with the Modbus RTU client. The full stacktrace looks like this:

````
panic: runtime error: slice bounds out of range

goroutine 6 [running]:
github.com/goburrow/modbus.(*rtuSerialTransporter).Send(0xc42011a008, 0xc42008e188, 0x8, 0x8, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/md/Projects/gosdm630/vendor/src/github.com/goburrow/modbus/rtuclient.go:153 +0x626
github.com/goburrow/modbus.(*client).send(0xc4200c7000, 0xc420440380, 0x2, 0xc42008e180, 0x4)
	/home/md/Projects/gosdm630/vendor/src/github.com/goburrow/modbus/client.go:443 +0xaf
github.com/goburrow/modbus.(*client).ReadInputRegisters(0xc4200c7000, 0xc400020020, 0xc42003ada0, 0x42c72e, 0x844d90, 0xa0faa0, 0x808313)
	/home/md/Projects/gosdm630/vendor/src/github.com/goburrow/modbus/client.go:143 +0xeb
github.com/gonium/gosdm630.(*ModbusEngine).retrieveOpCode(0xc420073290, 0x200416, 0xc42003aea8, 0x3, 0x3)
	/home/md/Projects/gosdm630/src/github.com/gonium/gosdm630/modbus.go:146 +0x93
github.com/gonium/gosdm630.(*ModbusEngine).Transform(0xc420073290, 0xc42006c660, 0xc42006c6c0, 0xc42006c720)
	/home/md/Projects/gosdm630/src/github.com/gonium/gosdm630/modbus.go:213 +0x14c
created by main.main.func1
	/home/md/Projects/gosdm630/src/github.com/gonium/gosdm630/cmd/sdm630_httpd/main.go:130 +0xa24
````

This occurs when the RTU connection is faulty, i.e. the line corrupts error messages. The corrupted messages are then shorter/longer as expected by the [functionFail code](https://github.com/goburrow/modbus/blob/master/rtuclient.go#L150). A length check before assuming 5 bytes to read should solve this.

I tried to capture more debug information (bytes on the wire), but so far I wasn't able to record a sample for this. Sorry!

nqv, 2017-07-20:
Thanks, I will fix this issue very soon.
Really sorry for the delay.

closed 2017-07-20 by nqv

gonium, 2017-07-25:
I just updated my project with your latest release, and connected my faulty device. I do not see the error, so in my case, your fix worked. Thank you!

commit 0d0a427b (2017-07-20): Fix reading more than the size of an exception response

Fixes #20