goburrow/modbus #23: ValueError: The slave is indicating an error. The response is: '\x01\x83\x02\xc0\xf1'

arsohel, 2017-08-07:
when i run the program its showing value error

import minimalmodbus
instrument = minimalmodbus.Instrument('COM3', 1) # port name, slave address (in decimal)
instrument.serial.baudrate = 9600
instrument.serial.bytesize = 8
instrument.serial.parity = minimalmodbus.serial.PARITY_EVEN
instrument.serial.stopbits = 1
instrument.serial.timeout = 1
instrument.debug = True
instrument.mode = minimalmodbus.MODE_RTU
print instrument
#l1_voltage = instrument.read_float(3911, functioncode=3, numberOfRegisters=2)
temperature = instrument.read_register(3911 , 2)
print temperature

laneschmidt, 2018-11-05:
?

andig, 2019-01-05:
This is a go library, you're using python?