goburrow/modbus #30: How to run?

agn-7, 2018-06-21:
I'm new in GoLang. 
I clone this project, how can I run a simple modbus/TCP holding register with these code files?

ArlenFuCN, 2018-06-27:
Hi ，you don't clone this.
With Golang,you just use "go get " to download this package and install it.
then,in your project,you have to set the "GoPath".
and next, you just "import " this package,you can use this package's method.
about "go get" of goland,I think you can google it.you will get lots of information about it.

agn-7, 2018-06-29:
Thanks, I set `$GOPATH=$HOME/go` in `.bashrc` and do `go get github.com/goburrow/modbus.git` in gopath location (~/go/) then I do `go install github.com/goburrow/modbus/`.
Then I create a a go project and use `go modbus` library in my code.

This is my "gopath" tree:

go/
├── src
│   ├── own_modbus
│   │   └── modbus.go
├── github.com
    │   └── goburrow
    │       ├── modbus
    │       │   ├──  . . .
              . . .

And this is my "own_modbus" code:

```
package main

import ("fmt"
		"github.com/goburrow/modbus"
)

func main() {
	cli := modbus.TCPClient("192.168.1.150:502")
	res, err := cli.ReadHoldingRegisters(4096, 1)
	fmt.Println(res)
	fmt.Println(err)

}
```

Is there any "go modbus" documentation and example such as [pymodbus](http://pymodbus.readthedocs.io/en/latest/source/example/modules.html)?

nqv, 2018-07-02:
There is an example: https://github.com/goburrow/modbus#usage
Docs: https://godoc.org/github.com/goburrow/modbus#Client