goburrow/modbus #34: Polling modbus TCP

mhemeryck, 2018-10-01:
HI; just a ticket for asking you a question.

I am currently working on a personal project requiring me to do a continuous polling loop on a modbus TCP server; see https://github.com/mhemeryck/modbridge for my project. The idea is to push any detected changes from the polling loop into an MQTT event. The use case I am looking at is having something PLC-like in a home automation setup.

For the modbus part, I heavily rely on your excellent modbus package. Some questions I have come across now:
* polling coils separately or as groups: my implementation solely relies on polling the modbus coils (for now). Initially, I figured to do a continuous polling of all the different coils one-by-one (in a goroutine). This works, but it seems rather overkill, considering in practice most of the coil addresses are sequential in nature thus allowing for this range to be polled and then interpreting these results. However, initial results do not seem to indicate that much of a performance gain. Question: **do you think it is better to poll the coils one-by-one or rather in a group?**
* TCP connections: if I am not mistaken, each of the calls would involve creating another TCP connection; **do you think there could be a gain by keeping the same TCP connection open for continuous polling**? Added to that**, what would be the best way to implement something like that**? (i.e. yet another implementation of the transporter interface?)
* architecture-wise: coming from python, I still have a lot to learn switching to golang for this project. I currently pass in a modbus client instance to my polling functions; would it make more sense to have something like a pointer to a singleton? **Any comments on the idiomatic way to handle this?**

Thanks in advance, and thanks again on this great package!

mhemeryck, 2018-10-02:
From looking the code (e.g. https://github.com/goburrow/modbus/blob/606c02f4eef527a1d4cf7d8733d5fd7ba34f91d8/tcpclient.go), the connection only closes after a time-out of 60s, so I assume it's not really required to manage the TCP connection any other way for my purpose, right?

nguyenthenguyen, 2018-10-02:
@mhemeryck You no need manage TCP connection. Connection closed when [Close method](https://github.com/goburrow/modbus/blob/606c02f4eef527a1d4cf7d8733d5fd7ba34f91d8/tcpclient.go#L232) called or after timeout when not have any activity by flag `mb.lastActivity`

mhemeryck, 2018-10-02:
Thanks for the quick feedback, this confirms what I found there :+1: 

That's actually nice, since it means I don't need to do any management on the TCP connection level. I would only need to make sure I would re-use the same client -- which I am already doing. I could perhaps find some speed-up in having multiple clients each doing their own polling (but then the server would need to be able to keep with that as well).

For the rest I guess I will need to do a more proper time profiling in my own package to see how I can further increase the reaction time.

mhemeryck, 2018-10-10:
Just noticed this ticket was still open; my question was pretty much answered by now.

For my setup, I did end up having each of my groups of coils to be polled have their own client instance. I still think it's probably not that "pretty", but as the client manages the connection itself, it makes sense I suppose.

closed 2018-10-10 by mhemeryck