goburrow/modbus #43: Working with multiple serial devices

pbdqpbdq, 2018-12-26:
Hi

I've successfully used the library to interact with one RS485 device, however working with multiple devices seems less than ideal.
Currently SlaveId is a property of the handler which needs to be adjusted every time. I think it'd make more sense to derive multiple Clients each with its own SlaveId from the handler and then be able to work with them without touching the handler. What do you think?

ibrt, 2019-09-08:
I have the same issue. Seems like an incorrect assumption in the way the abstraction is modeled, i.e. one serial port can have multiple devices connected to it on a bus. Other than that this library is great.

houaq, 2020-01-13:
same issue +1, maybe I should make a fork.

gaorongfu, 2021-03-16:
> same issue +1, maybe I should make a fork.

Hello! Have that fork been done?

maddie, 2021-03-16:
> > same issue +1, maybe I should make a fork.
> 
> Hello! Have that fork been done?

I have migrated to github.com/simonvetter/modbus

gaorongfu, 2021-03-16:
It's great! Thanks a lot !

houaq, 2021-03-16:
Yes, you can try it at houaq/modbus.

发自我的iPhone