goburrow/modbus #47: Error: "System cannot find the file specified." opening port when port number is high ex: "COM15"

Edgar242, 2019-06-06:
I have windows 10 and when I try to open serial port "COM15" I got:
"System cannot find the file specified." 

the issue is solved if I manually changed it to another closer port to COM1 for example: "COM2"

Edgar242, 2019-06-14:
Problem is solved here:

https://github.com/goburrow/serial/pull/13

alikxus, 2019-11-15:
Hello, Problem is not solved. I use com10 and i recieved - COM10 - The system cannot find the file specified. I need your help

Edgar242, 2019-11-15:
> Hello, Problem is not solved. I use com10 and i recieved - COM10 - The system cannot find the file specified. I need your help

Check this pull request, it has the changes need
https://github.com/goburrow/serial/pull/13

alikxus, 2019-11-18:
> > Hello, Problem is not solved. I use com10 and i recieved - COM10 - The system cannot find the file specified. I need your help
> 
> Check this pull request, it has the changes needed
> [goburrow/serial#13](https://github.com/goburrow/serial/pull/13)

Do I understand correctly? i need use this string, for example:
handler := modbus.NewRTUClientHandler("\\\\.\\com10")

Edgar242, 2019-11-18:
> Do I understand correctly? i need use this string, for example:
> handler := modbus.NewRTUClientHandler("\\.\com10")

Yes, I think name must be in uppercas: "\\.\COM10"
Can you try please?

alikxus, 2019-11-19:
thank you, editor deleted first 2 back slaches
[image]
this way work for me.

Edgar242, 2019-11-19:
> thank you, editor deleted first 2 back slaches
> [image]
> this way work for me.
> and it work without uppercas

Great, nice to know!

Tamilselvan-sakthivel, 2023-10-25:
i am using windows 7, if i the above mentioned method i am getting same error.

CreateFile("\\\\.\\COM2", GENERIC_READ | GENERIC_WRITE, 0, NULL, OPEN_EXISTING,0, NULL);

can anyone help me on this ?

Tamilselvan-sakthivel, 2023-10-25:
tried that too, but still same