goburrow/modbus #61: Read data out of SMA sunnyBoy

rogierlommers, 2020-06-25:
Hey, maybe a bit stupid question, but do you think it's possible to use this package to read data out of my sunnyboy inverter? According to the website it supports modbus. I also opened port 502 and by using `netcat` I indeed see some response on that port. But then what? What is my next approach in finding out what "registers" I should read to get the current generated solar energy?

wz2b, 2020-08-13:
i'm doing it.  what model inverter?  you need to find the right modbus map for your inverter.  They are not consistent across models.

I am using telegraf, which uses this project for modbus protocol support.

rogierlommers, 2020-08-16:
Thanks for your reply; I want to read a SMA Sunnyboy, model "Sunny Boy 3.6 1AV-41". Any idea how to proceed with this?