goburrow/modbus #63: Feature request: ability to set unit id on a request by request basis

wz2b, 2020-08-13:
I have a gateway with multiple "slave" (we should rename that "unit") devices attache.  Need to be able to specify slave addr per call, rather than changing the per connection SlaveID which works but raises potential concurrency problems.

shantanoo, 2022-02-14:
Just thinking out loud, would setting `handler.rtuPackager.SlaveId` just before the operation help?

alpr777, 2022-02-15:
i made it so for myself, maybe it will help
https://github.com/alpr777/modbus
https://github.com/alpr777/modbus/blob/main/example/main.go

pxnky, 2023-03-25:
> Just thinking out loud, would setting `handler.rtuPackager.SlaveId` / `handler.SlaveId` just before the operation help? https://github.com/grid-x/modbus/blob/master/rtuclient.go has `SetSlave` function.

Then you have to keep the `handler` around through all the rest of the code. This seems like a very basic feature, I don't understand why the assumption (seems to be) that you only set it once.